NVlabs/DREAM #10: How to create dataset of other type of robot using NDDS

hetolin, 2021-02-10:
Hi,
It is really great work and useful in real robot applications. 
Would you share the details on how to use the augmented NDDS tools to make the datasets? It seems that the provided NDDS tools cannot control the joints of the robot. 

Or could you please share the processing script or UE4 project files that may be easier to use, which may accelerate the process of making datasets using our own robots?

TontonTremblay, 2021-02-10:
We are currently switching out synthetic data pipeline to https://github.com/owl-project/NVISII, I would highly recommend you to look into it. We have examples with pybullet and exporting meta-data in the repo. I have been able to render robots as well using this tool and pybullet. Hopefully this helps.

hetolin, 2021-02-10:
NVISII is really a great tool that can generate photorealistic images. But could you share the example/python script that you used (using this tool and pybullet) to render robots?
 It seems NVISII lacks detailed documents for the moment, it may be a little difficult for a new learner to complete the synthetic data pipeline.

timeous, 2021-02-11:
Thanks for your kind remarks and question, @hetolin! I am sorry but at this time, we don't have plans for releasing the version of NDDS that was used to generate these datasets for DREAM that included robot joint control. Per @TontonTremblay, the current plan is to migrate synthetic data generation to NVISII.

TontonTremblay, 2021-02-12:
>  It may be a little difficult for a new learner to complete the synthetic data pipeline in a short time. 

I am sorry, at the moment I don't have a solution to be consumed quickly. But looking through the examples provided should help you out. Check 03 and latest one I added should give you a very good start.

hetolin, 2021-02-12:
Thank you very much. I will take it a try