NVlabs/DREAM #11: Doubt regarding the ADD Metric

b-yogesh, 2021-06-22:
Hello team,

I have a doubt regarding the way the ADD metric is computed in your code as I am unable to understand it.
In your paper, you mention that you refer to the ADD metric's implementation from https://arxiv.org/pdf/1711.00199.pdf, and here they show the formula of ADD.
[image]

But in your code (https://github.com/NVlabs/DREAM/blob/master/dream/geometric_vision.py#L199), here you are subtracting two 3D points in different coordinate systems (since you transform the ground truth to another coordinate by using the transform matrix, and then subtract the transformed version from the original). How does that work and what is the intuition behind it?

I have trouble understanding this, and I would be really glad if you can help me in clearing out this doubt.

Thank you :)

tabula-rosa, 2021-09-14:
Hi Yogesh! Thanks for your interest in our work. If my explanation below isn't clear, please let me know and I will try to re-explain.

Our implementation of ADD looks different than that formula, but mathematically, it should be equivalent. The short answer is that our formulation is simpler because of how we construct the problem: we only need to apply one SE(3) transformation (the transformation that arises between ground truth [g.t.] and est. camera frames), rather than two SE(3) transformations (between g.t. camera and robot, and between est. camera frame and robot) that the formula above suggests. This is because the camera pose that gets input to `add_from_pose` is the pose between the g.t. and est. camera frames, so it's less implementation to use this directly while still giving an equivalent result.

Here's the longer answer:

Consider that the expression "(Rx + T)" in the above formula is applying the SE(3) transformation that arises from rotation R and translation T to the point x.

For our case, let the ground truth camera pose with respect to ("wrt") the robot base be defined by R and T. And let x be the position of a specific keypoint wrt the robot base. The expression "(Rx + T)" is taking the keypoint expressed wrt the robot base and transforming it, so it is now expressed wrt the g.t. camera frame. (These are the points described by `keypoint_positions_wrt_cam_gt` in the code.)

Now, consider that R_est and T_est in the above formula are from the SE(3) transformation that arises from the estimated camera pose wrt the robot base. From the logic above, "(R_est x + T_est)" is the same keypoint as before, but now expressed wrt the estimated (not ground truth) camera frame. We might call this `keypoint_positions_wrt_cam_est`, if we needed a variable name for it.

However, we aren't given `keypoint_positions_wrt_cam_est` -- at least, not directly. But, we can calculate this from other information we do have.

In this function `add_from_pose`, we are given the pose between the ground truth and estimated camera frames. (This is the low-level output from PnP, `translation`, and `quaternion`). We can form a SE(3) transformation from this pose, and apply it to the keypoints expressed wrt to the g.t. camera frame (`keypoint_positions_wrt_cam_gt`). This gives us the keypoints expressed wrt the estimated camera frame -- or, `kp_pos_aligned` as it is called in the code.

Now, we have both sets of information we need (keypoints wrt g.t. camera frame, keypoints wrt est. camera frame). We subtract these, take the L2 norm, and calculate the mean to get ADD.

tabula-rosa, 2021-09-15:
In case it helps, here is the math I am describing above. I am using the mathematical notation where T represents a SE(3) transform arising from rotation R and translation T, where the superscript and subscript of T indicates the coordinate frames that the transform relates. There are three coordinate frames involved: B (robot base), C (ground truth camera), C_est (estimated camera).

[image]

closed 2021-09-22 by tabula-rosa