NVlabs/DREAM #13: TF Exception: "torso" passed to lookupTransform argument target_frame does not exist.

JaniC-WPI, 2022-04-14:
Hello,
To begin with I want to thank you for this awesome framework. I am trying to work with this framework and ran the ROS node, by running the following script from terminal 
python scripts/launch_dream_ros.py -i trained_models/baxter_dream_vgg_q.pth -b torso -v
However I got the following issue 
TF Exception: "torso" passed to lookupTransform argument target_frame does not exist. 

I tried it with the panda_dream_resnet_h.pth training model and with base frame as 'Base' and 'panda_link0'. Both gave me the same issue as above. Please help me undertsand what I am doing wrong. And how to rectify the issue.

Cheers,
Jani

tabula-rosa, 2022-04-15:
Hi Jani! Thanks for your interest in our work! Sure, I'd be happy to help. I'm wondering if it's due to some differences in TF naming conventions for your setup.

To start, would you mind running the following command when you receive that error: `rosrun rqt_tf_tree rqt_tf_tree`
This should yield a GUI with the TF tree visualized. Please save the resulting TF tree (there should be a "save" in the upper right corner) and upload it here. (Or, you can email it to me directly at [tabithalee@cmu.edu](mailto:tabithalee@cmu.edu) if that's easier for you.) Then, I can take a look to help debug what's going on. Thank you!

JaniC-WPI, 2022-04-15:
> Hi Jani! Thanks for your interest in our work! Sure, I'd be happy to help. I'm wondering if it's due to some differences in TF naming conventions for your setup.
>
> To start, would you mind running [...]

JaniC-WPI, 2022-04-15:
Hello Tabitha,

Thank you for replying to my issue. However, I was just able to resolve the issue. And it is not from your side. I was fetching the wrong robot description. I think this issue also should not be in the Issue list here. Please let me know if I should delete this.

Regards,
Jani

closed 2022-04-15 by JaniC-WPI

tabula-rosa, 2022-04-15:
Great! I'm glad that the issue has been resolved! Re: deleting this issue, as far as I am concerned, it is no problem to keep it as a closed issue. But if you'd like to delete it, you are welcome to do that.

All my best,
Tabitha