NVlabs/DREAM #14: How to install DREAM globally

JaniC-WPI, 2022-04-15:
Hello All,

This is not an issue but a query on installation. I want to install pip globally in my machine, so that I can use it as numpy and such. Is that possible? 
So far I have cloned the repo first and then installed the package as instructed using the 'pip install . -r requirements.txt' inside the folder crated during cloning. However I need to use the package with a ROS package and was wondering is that could be possible. 
Please help.

Reagrds,
Jani

tabula-rosa, 2022-04-22:
Hi Jani! Thanks for the question and interest in DREAM!

For more information, the `pip install . -r requirements.txt' command will install DREAM to the virtual environment that is currently active. If I'm not mistaken, if no virtual environment is sourced, then the `pip install` command would be to the base environment --- which is what is used when there is no specific virtual environment sourced.

By installing globally, do you mean having it available across all virtual environments? There might be a way to do this, but I am not certain and I haven't tried, so I'm afraid I can't help too much here.

Regarding your use case of using both DREAM and ROS --- yes, this should generally be possible, although it may depend on what capabilities of ROS you are looking to use. DREAM was tested on both Python 2.7 and 3.6, and tested with Python 2.7 with ROS Melodic, and so it should be possible to have both DREAM and ROS working within one virtual environment. For this, you might need to have a Python 2.7 virtual environment, and source ROS from there.

Generally, what I do is only source ROS when I need to use it (even though the ROS installation instructions say to source ROS in your `~/.bashrc` file. For example, in my `~/.bashrc` file, I have:
```
alias source_ros='source /opt/ros/melodic/setup.bash'
```
which means that I would use a Python 2.7 virtual environments as usual, but only source ROS when I need to (using this alias). I'm not sure setting up your environments in this way would be necessary to use DREAM and ROS, but this is what has worked for me.

ROS Melodic was released primarily with Python 2.7 in mind, although some modules work with Python 3.6. In case there are any issues with incompatibilities, another option is to maintain two separate virtual environments --- one with Python 2.7 (primarily ROS) and one with Python 3.6 (primarily DREAM). The ROS code on the Python 3.6 environment would primarily be for message handling --- taking the ROS messages and converting them in a way to use DREAM. So, the ROS message passing helps bridge the Python 2.7 and 3.6 environments.

I hope that helps, and in case you need any additional information, please let me know!

tabula-rosa, 2022-06-13:
I'll go ahead and close this issue for now, but in case you have additional challenges, please comment and I'd be happy to re-open the issue to help troubleshoot!