NVlabs/DREAM #16: Is it simtrack necessary?

qdzr, 2022-06-12:
Hello,
Simtrack is used to refine your pose. However, simtrack seems not to be maintained. And unfortunately simtrack needs large gpu memory. I want to know if the error will be large without simtrack to refine the pose.  Or is there an alternative to simtrack?

tabula-rosa, 2022-06-13:
Hello and thank you for your interest in DREAM!

Unfortunately I'm afraid I don't understand your question, since I don't think DREAM uses simtrack. Would you please provide more information or a reference to a file in the repository so that I can help with your inquiry? Thank you!

tabula-rosa, 2022-06-18:
Although I'm not an expert in that paper (and I would defer to @TontonTremblay if he has any suggestions here), I would say that SimTrack is not a strict requirement, only a useful capability for pose refinement. In principle, any pose refinement algorithm or software module should work. We have observed that sometimes there can be fluctuation of the pose output from DREAM, which depends on factors such as environmental lighting, so not all users would experience it. But this is where the pose refinement would be useful, especially if you wanted to also move the camera while DREAM is running.

For alternatives, one option could be modifying the multi-frame mode we offer in the DREAM ROS node. At present, the multi-frame mode works with a manual capture. But, one option is modifying this capability into an automatic queue. That is, every X seconds (X roughly between 0.1 to 2 seconds) the frame is automatically captured. Then, once Y frames (Y roughly between 5 and 20) are collected, you'd start dropping the oldest frame each time a new frame is captured. There is a buffer in the DREAM ROS node which is where the frames would be collected, and PnP is run on all of these frames. So, this option would turn the manual multi-frame mode into an automatic pose refinement capability that functions as a low-pass filter.

qdzr, 2022-06-25:
Thanks for your reply. And your advice helps me a lot. I will close the issue now.

closed 2022-06-25 by qdzr