NVlabs/DREAM #21: Potential Bug : When comparing multiple peaks in a belief map

ThakurSarveshGit, 2022-10-11:
Hi,

On this line : https://github.com/NVlabs/DREAM/blob/d71b1760b02f831f613229838e0ae9ea39340100/dream/network.py#L556
Peaks are compared based on y-coordinate instead of scores, which is not making sense. Please clarify.

Proposed Code: Line 554-557 in dream/network.py:
```
# Try to use the belief map scores
peak_sorted_by_score = sorted(
    peak, key=lambda x: x[1], reverse=True
)
```
Also, what makes the network output two peaks for a belief map? During training, shouldn't it have learned to output only one belief area for a keypoint? Looking forward to your response. Thanks!

ThakurSarveshGit, 2022-10-12:
@tabula-rosa Thanks for the response. I am using DREAM for inference on a Kuka iiwa7 arm in a drake simulator. When the arm is extended completely(straight up), I see that keypoint suffers. I am not sure if such images(where arm is completely extended) or occluded) were part of the dataset while training. If I were to finetune the model to work well on such scenarios too, what should I keep in mind while creating the dataset(of kuka arm in drake simulation environment)? Thank you!

ThakurSarveshGit, 2022-10-12:
> That's a great use case for using DREAM! If I remember correctly, I'm not
> sure the Kuka training data has many examples of the arm completely
> extended straight upward, so the performance drop you're seeing is likely
> due to a mismatch in train/test data. Fine tuning should be able to help,
> although I unfortunately can't provide more specific recommendations as I
> personally have not fine tuned the network. I would probably start with a
> smaller dataset that consists of a mix of training data examples and the
> test data examples and go from there. Best of luck!

Thanks for the response!!

Regards,
Sarvesh

tabula-rosa, 2022-10-12:
Your welcome! The bug fix for the belief map peak scores is now in the master branch, so I'll go ahead and close this issue. Thanks and best of luck!

closed 2022-10-12 by tabula-rosa