NVlabs/DREAM #24: Questions regarding the meaning and usage of the "sim_states" in json files of the synthetic dataset

Oliverbansk, 2023-02-18:
Hello! This is really fantastic work and I enjoyed reading your code and your paper. Right now I have some questions regarding the meaning and usage of the "sim_states" in json files of your synthetic dataset.  The part is pasted below:
[image]
Can you elaborate further on what "joints" here means? I'm guessing that in the "joints" list, the numbers that "position" referred to, actually represents the joint angle(value/configuration). Is that right?
We are working on a pipeline that takes in joint angles/values and computes 3D keypoints location, if the "position" isn't the joint angle , we would like to know where we can get those values ( I think for each photo of the robot in the dataset there should be a set of joint values that matches the robot ).
Thanks!!

ThakurSarveshGit, 2023-02-18:
> Can you elaborate further on what "joints" here means? I'm guessing that in the "joints" list, the numbers that "position" referred to, actually represents the joint angle(value/configuration). Is that right?
Yes

tabula-rosa, 2023-02-18:
Thanks for your kind remarks and interest in DREAM, Oliver!

As Sarvesh mentioned, my understanding is that these positions under "joints" are indeed the joint angles, measured in radians.

For more information, the data within the "sim_state" fields were not used during algorithm development, and I no longer have access to the simulation engine that generated these files, so I unfortunately won't be able to provide much detailed insight into the sim_state data. For example, the `sim_state:links` fields appear to describe the poses of each link's coordinate frame, but I am unsure whether the convention is with respect to the base link, the camera, or the world. The units for the link positions are probably meters. All that being said, you are welcome to try using the "sim_state" data! If you have success, please feel free to report back and share for other researchers who may want to use these data as well.

Oliverbansk, 2023-02-20:
Thank you! We will give this a try! Actually I have another question regarding the synthetic dataset. It's the "pose_transform" matrix! 
[image]
I assume that the "pose_transform" matrix is the base-to-camera transformation matrix, and I think the "location" and "quaternion_xyzw" above “pose_transform" matrix can be deducted from the matrix. (quaternion_xyzw for rotation)
But what's confusing me is that "quaternion_xyzw" doesn't seem to match the “pose_transform" matrix.
I converted the "quaternion_xyzw" to a 3x3 rotational matrix using some robot fk calculation code, as pasted below:
[image]
It shows:
[image]
Based on 3D geometry maths, I think the 3x3 rotation matrix converted from the quaternion should be identical to the upper-left 3x3 area of the 4x4 Transformation matrix which in this case, is  the “pose_transform" matrix. But the result says otherwise as you can see the 3x3 matrix contains inconsistent values with those of the “pose_transform" matrix in my 1st screenshot.

So I'm wondering if the "quaternion_xyzw" doesn't describe the same rotation as the "pose_transfrom" matrix does? In the meantime I will try some new conversion code but I suspect that it might be posing same problem as this.

Thanks!!

tabula-rosa, 2023-02-22:
After checking this myself, I can confirm that the `objects:quaternion_xyzw` field doesn't appear to be consistent with the `objects:pose_transform` matrix. Another complication is that the `camera_data:quaternion_xyzw_worldframe` entry seems quite similar to that of `objects:quaternion_xyzw` although with some elements in a different order. So, I'm afraid I can't provide much insight as to what coordinate frames/conventions were used in calculating these orientations. I don't have access to the simulator that generated these files, so I can't check myself. These fields were not used for our algorithm development, so it's possible the convention used was uncommon, unintuitive, or possibly buggy. To further complicate matters, I remember that the simulator that was used (Unreal Engine) actually uses a _left-handed_ coordinate frame convention internally, so if these fields are using that convention, it could explain why the orientations aren't consistent. If there weren't any bugs in the calculation of these fields, then I would suspect there would be a set of transformations/conventions that would cause the entire data to be consistent. But, again, I can't confirm whether these fields were calculated bug-free. I wish I had a better answer, but that's the extent of my knowledge. Best wishes and let us know how it goes!

Oliverbansk, 2023-02-23:
Thank you for your help!! I'm using the"pose_transform" matrix, it seems to be what I want. For now I'll just leave the quaternion behind. Thanksxxxxx!

closed 2023-03-05 by Oliverbansk