NVlabs/DREAM #26: Would you like to share your modified panda URDF?

Oliverbansk, 2023-02-28:
Hello! Thank you for your explanation before! Really helped a lot!
Now I've encountered another coordinates offset error when using the Panda robot, but it doesn't seem similar to the error I've encountered before. I wonder maybe it has something to do with your modification on Panda URDF. 
I'm wondering if it will be okay for you to give more information regarding your change to the original URDF. 
Thank you so much!!!!!!!!

tabula-rosa, 2023-02-28:
Hi Shikun! Thanks for your continued interest in DREAM!

Yes, the Panda URDF was modified to change the coordinate frame poses of the links, such that there would be a keypoint at each of the actual joint locations. This led to training a Panda network that detected 9 keypoints, which is what is shown in Figure 5 in our paper.

With the original Panda URDF from Franka Emika, the link coordinate frames were defined such that there were only 7 unique keypoints. For example, in the original URDF, panda_link1 and panda_link2 coordinate frames share the same origin in 3D space, so their projected location in 2D would be the same, leading to one keypoint, not two. The same is also true for panda_link5 and panda_link6.

We chose not to release this modified URDF, as we wanted our released networks to be consistent with the URDF that would be more widely available, and we didn't want to complicate matters. That's why the released Franka networks detect 7 keypoints; it assumes the conventions from the original URDF.

Unfortunately, I'm unsure if the modified URDF is still available. I do not have the file, and I no longer have access to documents from my internship, when this project was conducted. However, the modified URDF should be able to be re-created. If you were looking to do this, I would suggest redefining the links/joints in the original URDF so that spatially there are no overlaps in the link coordinate frames. This would need to be done twice: once for panda_link1/panda_link2, and again for panda_link5/panda_link6. That's how the modified URDF was obtained in the first place. I hope that helps!

Oliverbansk, 2023-03-03:
Thank you a lot!!! So apart from link1 and link5, other links should remain the same with the original URDF, including the link_coordinates. This helps a lot!!!

closed 2023-03-05 by Oliverbansk