NVlabs/DREAM #28: Robot Pose without ROS

rlee3359, 2023-04-06:
Hi, thanks for the cool repository!
I'm getting some nice looking keypoints from the inference script on images from my Panda setup, and so I'd like to try getting the calibration transform/pose. Is there a way to obtain the transform, by inputting an image, the joint positions and camera intrinsics?

The ROS node looks great but I'm not using ROS to control my robot, and it seems to rely on tfs from all joints. Could I get some help setting a simple inference script without ROS?

Thanks so much!

tabula-rosa, 2023-04-06:
Hi Rob! Thanks for your interest in DREAM!

If I understand your question, it seems like you're looking for a way to get the camera-to-robot transform without using ROS. Yes, this should definitely be possible. The major components are available in the ROS script and can be repurposed to provide the transform without ROS.

As you mention, the largest change will be for getting the 3D keypoint positions needed for PnP. You are correct that this is done in the ROS script with TFs/`lookup_transform`. To do this step without ROS, you can use any forward kinematics calculator, that given the Panda model and the joint positions, returns the kinematics of the robot. I have used the [Robotics Toolbox](https://github.com/petercorke/robotics-toolbox-python) before for inverse kinematics with the Panda, so it should also be able to do forward kinematics too. 

With that, the general procedure for such a script is:
1. Getting the 2D keypoints from the image ([link](https://github.com/NVlabs/DREAM/blob/dd0a047f3bc1462fbca46a0895524761fd6f3ebe/scripts/launch_dream_ros.py#L256))
2. Getting the 3D keypoints from forward kinematics using joint positions
3. Running PnP to get the transform using the 2D keypoints, 3D keypoints, and intrinsics ([link](https://github.com/NVlabs/DREAM/blob/dd0a047f3bc1462fbca46a0895524761fd6f3ebe/scripts/launch_dream_ros.py#L450))

Lastly, the code used to generate those visualizations is in the codebase. The left image is from the ROS topic "/dream/keypoint_belief_overlay" and the right image is from the ROS topic "/dream/keypoint_frame_overlay". The keypoint belief overlay visualization code starts on [line 317 of the ROS script](https://github.com/NVlabs/DREAM/blob/dd0a047f3bc1462fbca46a0895524761fd6f3ebe/scripts/launch_dream_ros.py#L317). The keypoint frame overlay visualization code starts on [line 505 of the ROS script](https://github.com/NVlabs/DREAM/blob/dd0a047f3bc1462fbca46a0895524761fd6f3ebe/scripts/launch_dream_ros.py#L505). Both use image processing functions provided in the `image_proc` module of the DREAM library. The keypoint frame overlay code might be harder to adapt since it also uses TFs to get the coordinate frame of each keypoint, but in principle this can also be provided by a forward kinematics module.

Hope that helps! Let us know if you have any further questions!

rlee3359, 2023-04-06:
Thanks so much, that's very helpful.

After some investigation, I found that the CtRNet repository uses the Robotics Toolbox to do this, exactly as you described  [here](https://github.com/jingpeilu/CtRNet-robot-pose-estimation/blob/7289b55033a418128448bea7a6835bae3e40dfaa/models/robot_arm.py#L53). I should be able to adapt that for step 2. The ROS node uses the the arg `args.base_frame` as the base frame for the TF lookup, does this base frame matter for PnP? Can I just get the 3d points with FK and keep them with respect to the robot base joint?

Lastly, I understand the URDF was modified for the paper, but based on this issue #26 , I should have no problems with just using the standard Panda URDF right?

Thanks for the info about the visualisations, I missed that in the codebase!

Appreciate the help!

tabula-rosa, 2023-04-07:
Great! Yes, that code in the CtRNet repository should be exactly what you need.

To answer your question about `args.base_frame`, in practice this is the coordinate frame of the robot base joint. So, the nominal coordinate frame of the 3D keypoints from FK will most likely already be in the correct coordinate frame.

About the URDF, the DREAM results for the Panda in the paper were using the original URDF but only using 7 of the possible 9 keypoints (assuming keypoints are located at joints). We later modified the URDF to use all 9 keypoints, re-trained the model, and produced Figure 5 in the paper. The DREAM network that was released for open-source assumes the original Panda URDF, so if that's what you are using, it should work. Just note there will be no keypoint for link1 and link5.

Best wishes for your script! If you decide to open-source it, you are quite welcome to post the link in this issue for others to use.

If you have any more question, please let us know!

tabula-rosa, 2023-08-26:
Since there haven't been any further questions, I'll go ahead and close this issue as resolved for now. Please let us know if you have any additional questions and we can re-open this issue to discuss further!

closed 2023-08-26 by tabula-rosa