NVlabs/DREAM #29: Does your ADD metric include the keypoints that are out of the image?

Oliverbansk, 2023-08-25:
Hi! This is really fantastic work and I enjoyed reading your paper. Actually I would like to know whether your ADD metric includes the keypoints that aren't in the image / camera viewing frustum.

Reading your paper and your code, I know that you guys discarded those images that have less than 4 visible keypoints because those images cannot go through a pnp process. But did your ADD metric include the keypoints that are out of the image?  Your paper said that you used the keypoints in the camera viewing frustum to calculate the ADD.

But the running result of your code seems that you guys included all the keypoints whether or not it's captured in the image (didn't managed to locate something simalar to "valid_mask" that represent keypoints in/out of the image).

I'm wondering if you can share that information directly with me, if my understanding on your paper or my running is wrong, I'd be really appreciated if you can point it out for me!

Thanks~~

Oliverbansk, 2023-08-26:
Sorry about the abrupt question, we now understood your implementation details and found the in-frame / out-of-frame keypoints reported in your summary. Your report points out the FP/TP on the keypoints prediction, which is very useful to us.

TontonTremblay, 2023-08-26:
Thank you for the update, I was going to look at the implementation over today, glad you figured it out.

tabula-rosa, 2024-02-16:
It seems like this issue is resolved, so I'll go ahead and close it for now. If there are any concerns or further questions, please feel free to reopen this issue so we can help!

closed 2024-02-16 by tabula-rosa