NVlabs/DREAM #31: Synthetic data annotations

thaiph99, 2024-02-16:
Hi, thanks for a nice repository.
I have a question about Panda synthetic data keypoints annotation:
In "Note on Panda keypoints" part, You said you modified the URDF to place the keypoints at the actual joint locations.
So what is the actual joint location, you can explain more about that, and share the URDF change file ?
Many thanks again.

tabula-rosa, 2024-02-16:
Good morning Thai! Thanks for your interest in our work and for the kind words about our repository!

Yes, you are correct in that the Franka Panda URDF was modified so that there is a keypoint placed at each joint and in total, 9 keypoints. This was due to the specification of the link coordinate frames in the original Panda URDF, where some link coordinate origins coincide at the same point in 3D space, leading to only 7 unique keypoints.

Although we do not have plans to release the modified URDF, it would be possible to update the URDF in such a way to reproduce this change. Please see our response to this issue here, which discusses the problem in more detail and recommends how to update the URDF:
https://github.com/NVlabs/DREAM/issues/26#issuecomment-1449063576

Please let us know if you have additional questions!

thaiph99, 2024-02-20:
@tabula-rosa Great, Thanks for replying. I understand a little bit of it, but I am having some problems generating data from the simulation environment (Isaacgym). So can you give me any tutorials for this work?

thaiph99, 2024-02-21:
> Hey un related to trying to reproduce this work, I have worked on randomly generating animated robots. https://github.com/TontonTremblay/blender_robot_animation The data it exports has the keypoints you need. You will have to pipe things a little bit, but you should be able to train on this data.
> 
> Here are some samples: https://drive.google.com/drive/folders/1MxatSrvoFA7MnvP1MyfHdVxhn9P47QAt?usp=drive_link

That is a great repos, I will read it, thanks for your reply.

closed 2024-02-21 by thaiph99